CTXz/stm32f1-picopwner #2: All I get is "Debug probe already connected, please disconnect the debug probe for now"

deividAlfa, 2023-09-03:
Hi, I'm probably missing something stupid, but I cannot get it to work.

- Flashed the pico with the latest release.
- Using Debian 20 VM, it recognizes the Pico and creates /dev/ttyACM0.
- Extracted the attack binaries into ./target folder.
- Python version: ```3.7.3```.
- Ran ```pip3 install pySerial```, it sucessfully installed the module.
- Installed OpenOCD 0.10.0 from apt.

I'm executing ```python3 dump.py -p /dev/ttyACM0 -o dump.bin```

All I get is:
```
Instructions:

1. Flash the attack firmware to the Pi Pico

2. Connect the Pi Pico to the STM32F1 target as follows (left Pico, right STM):
    GND -> GND      
     0  -> USARTx_RX
     1  -> USARTx_TX
     2  -> VDD      
     4  -> NRST     
     5  -> BOOT0    

Where:
     USART1_RX = PA10, USART1_TX = PA9
     USART2_RX = PA3,  USART2_TX = PA2
     USART3_RX = PB11, USART3_TX = PB10

3. Follow further instructions provided by this script
For more detailed steps, see the README.md file.

Debug probe already connected, please disconnect the debug probe for now
```

I'm getting this message no matter what, even if /dev/ttyACM0 doesn't exist or specifying a non-existing port like ```/dev/asdfg```.

deividAlfa, 2023-09-03:
Answering myself: OpenOCD doesn't contain any file called ```stlink.cfg```.
I edited ```dump.py```, replacing all the entries containing```stlink.cfg``` with  ```stlink-v2-1.cfg```.

deividAlfa, 2023-09-03:
More issues.
Case correction around dump.py line 249:
```
    for line in result.stderr.splitlines():
        if "readout protection on" in line:
            return True
        elif "readout protection off" in line:
            return False
```

Should be:
```
    for line in result.stderr.splitlines():
        if "Readout Protection On" in line:
            return True
        elif "Readout Protection Off" in line:
            return False
```

deividAlfa, 2023-09-03:
And finally, I got it working!

- Connect the pico first so it takes the ACM0 port (The ST-Link will create another ACM port).
- Just in case run ```sudo dmesg``` and check the port number.
- Connect the ST-Link (If it's the serial port variant, will use ACM1), but don't connect it to the target SWD port.
- Remove the Pico (The script wants it freshly connected during execution).
- Run the script: ```python3 dump.py -p /dev/ttyACM0 -o dump.bin```.<br>

> Waiting for Pi Pico to be connected... (Looking for /dev/ttyACM0)

Connect the pico now. Run ```sudo chmod 777 /dev/ttyACM0``` in another window.

> Device connected to serial port /dev/ttyACM0
> Waiting for debug probe to be connected...

Connect the ST-Link SWD port to the target.

> Debug probe connected to STM32F1 target
> STM32F1 target is confirmed to be read protected
> Detected SRAM entry point offset: 0x108 (0x20000108)
> Please select the USART used by the STM32F1 target to dump firmware
> 1: USART1 - RX: PA10 TX: PA9)
> 2: USART2 - RX: PA3  TX: PA2)
> 3: USART3 - RX: PB11 TX: PB10)
> Enter 1, 2 or 3: 1

Choose the firmware depending on the stm32 pins you connected the pico's serial port.

> Press enter to load the target exploit firmware to the SRAM

Just that. Press enter.

> Target firmware loaded to the SRAM
> Waiting for debug probe to be disconnected...
> Warning: Disconnect the debug probe from the target, not just the host USB port!

Disconnect the ST-Link SWD port from the target, no need to remove the USB.

> Debug probe disconnected from STM32F1 target
> 
> Attack ready
> Press enter to start dumping firmware

Again, just that. Press enter.

> (Lots of hex data)

After 1 minute or so: 
> Target has stopped sending data, assuming dump is complete
> Dumped 65536 bytes
> Output saved to dump.bin

closed 2023-09-06 by deividAlfa

CTXz, 2023-09-07:
Ah, I was aware of the differences between older and newer OpenOCD versions but thought I had tested the script on 0.10 successfully. Turns out I likely didn't.

I'll specify 0.12 as the min version from now on and add a comment referring to this issue for users that wish to use 0.10. 

Glad you got it working!

deividAlfa, 2023-09-07:
For the the differences in ouput case, I think the solution would be to do a [case insensitive string compare](https://www.geeksforgeeks.org/case-insensitive-string-comparison-in-python/).

And for the stlink config file differences, should't be too hard to test if openocd likes "stlink.cfg" by checking the output.

> openocd -f interface/stlink.cfg

Openocd won't launch as there's no target, but we will be able to scan the ouput for "Can't find interface/stlink.cfg".
If so, then try: 
> openocd -f interface/stlink-v2-1.cfg