CTXz/stm32f1-picopwner #11: Failure to load target firmware to SRAM

sohail330x, 2023-09-29:
Hi, I'm not sure where I'm going wrong, was testing this on a Blue pill board- followed all directions and it goes smoothly until after crossing USART 1,2 or 3, when loading target firmware to SRAM, I get -

Exception: Failed to load target firmware to SRAM
openocd output: Error: couldn't open /c/Users/ME/Desktop/stm32f1hack/target/target_108_usart1.bin

Anyone able to help me with this?

CTXz, 2023-10-01:
Did you actually build or download the target firmware binaries? Doe `/c/Users/ME/Desktop/stm32f1hack/target/target_108_usart1.bin` exist?

sohail330x, 2023-10-01:
Hi , I used the ready made target firmware and the file path did exist- I found it was a software problem! I was running on Windows with all the prerequisites installed (openocd etc) 

I am now attempting on Linux and I managed to get further- I can successfully load target firmware but I get the result at the end: time-out: no data received from target 

All wiring has been checked and double checked and it's stumping me- tested for bluepill (target) power using st link without 3.3v and gnd and it connects so board is powered by Pico can't work out why it won't work

CTXz, 2023-10-01:
I've just published a new release that fixes some reset line related voltage issues (See https://github.com/CTXz/stm32f1-picopwner/releases/tag/a1.2_t1.2 for more details). Try to download the new attack and target firmware and report if that has perhaps helped.

sohail330x, 2023-10-02:
Hi thank you very much, I have tried the new releases- same issue : timeout: No data received from target 

One thing I have noticed- on the blue pill the red light for power is illuminated during the entire process, even after I remove debug probe (st link) from target. Once I click enter to start dumping firmware, this red power indicator turns off? Could this help to narrow down what I'm doing wrong?

Only power cycling the pico restarts the red power indicator on the bluepill so I think the issue I am getting is possible from the pico/ attack firmware?

JimKnopfIoT, 2023-10-02:
Use wires as short as possible. I had timeouts with 30 cm Dupont wires (i called it antennas). When using 7 cm wires directly soldered to pinheaders, i had 100 % success rate. If available, check signals on oscilloscope.
Check Boot0 pin, NRST, TX from bluepill board.
The first image show start of reading with 30 cm wires.
[image]

The second image show the start of reading with 7 cm wires (used different colors for signals).
[image]

CTXz, 2023-10-02:
> Hi thank you very much, I have tried the new releases- same issue : timeout: No data received from target
>
> One thing I have noticed- on the blue pill the red light for power is illuminated during [...]

This likely implies that the attack firmware is strugging to get past the following loop:

```c++
	// Drop the power
	gpio_put(LED_PIN, 1);
	gpio_put(POWER_PIN, 0);

	// Wait for reset to go low
	while (gpio_get(RESET_PIN)) {
		tight_loop_contents();
	}

	// Immediately re-enable power
	gpio_put(POWER_PIN, 1);
```

The pico will shut down power to the STM32F1 and re-enable it as soon as the reset pin has been read low. Ensure that the reset pin is really connected to GPIO4 (Pin 6, see below).

[image]

If you're confident the wiring is correct, try to capture an oscilloscope reading of GPIO4 and the RST line.

sohail330x, 2023-10-02:
Hi thanks for your reply,

Weirdly I went back to the previous 'dump.py' script and it is no longer doing the 2-3 second pause between powering off and on after clicking enter to start firmware extraction. So it seems to be power cycling properly now but still 'no data'

CTXz, 2023-10-02:
I'm afraid without oscilloscope readings or at least an overview of your set-up I can't really help out any further here

CTXz, 2023-11-05:
Since this issue has been inactive for longer than a month now, I will close it.
I will gladly re-open it if necessary.

closed 2023-11-05 by CTXz