CTXz/stm32f1-picopwner #12: Bluepill STM32F103 - waiting for Pico...

Aveneid, 2023-12-13:
Hi,
I tried to read read-protected Bluepill with Pico Pwner. I got all connections properly set and firmware prepared.
After running script i only got this:
[image]

Restaring Pico with RUN pin or reconnecting USB does not work.

I've checked common issues in readme but this one was not there.

Am I doing something wrong? 
Images:
Bluepill: (https://imgur.com/egfpyQL)[https://imgur.com/egfpyQL] (https://imgur.com/J13AtL2)[https://imgur.com/J13AtL2] (https://imgur.com/KBCYHS4)[https://imgur.com/KBCYHS4]
Pi Pico: (https://imgur.com/1HCmUFY)[https://imgur.com/1HCmUFY]  (https://imgur.com/vRPP7AO)[https://imgur.com/vRPP7AO] (https://imgur.com/qiAk7io)[https://imgur.com/qiAk7io]

Tried all versions with no luck.

CTXz, 2023-12-15:
Hi,

After plugging in your pico, can you confirm that `/dev/ttyACM0` exists?

Could you post the last 50 or so lines of `dmesg` after plugging in the pi pico?

Thanks

Aveneid, 2023-12-15:
Im running it on physical machine, after plugging Pico ACM0 exists but no communication between PC and board.

I tried to run it under WSL but i had issues with COM port forwarding from Windows 10 machine, maybe this was not the case?

dmesg log: https://pastebin.com/9h06Cxme

CTXz, 2023-12-15:
Alright, I recommend you continue working on a native/hardware setup since I have not tested the script under WSL.

Could you run the following python script and post the output:

```
from serial import Serial, SerialException

ser = Serial('/dev/ttyACM0', 9600, timeout=0.5)
print(ser)
```

I suspect that `Serial()` is throwing an exception related to the issue but it is simply interpreted as a unconnected device due to the way the `wait_serial_connect()` function has been set-up.

Aveneid, 2023-12-15:
I tried this script and got error `Permission denied`. 
Searched in Google about this error and found post on stackoverflow with solution as so:

https://stackoverflow.com/a/27886201/6369954
> 01. Alter the permissions on /dev/ttyACM0 so that world has read and write priviliges (something you may not want to do) - although you may find they reset each time the device is plugged in eg:
> 
> sudo chmod 666 /dev/ttyACM0 

Tried again with script as You wrote and it worked: https://pastebin.com/ZnusbymU

In my case issue is that `/dev/ttyACM0` is set read-write for root user only, other user cannot interact with this device.

After setting ACM0 to 666 I've managed to connect to Pico but stuck with `Waiting for debug probe to be connected...`

CTXz, 2023-12-15:
An obvious question but have you connected the debug probe?

What's the output if you run the following command after your debug probe has been connected:

```
openocd -f interface/stlink.cfg -f target/stm32f1x.cfg
```

What's your openocd version (run `openocd -v`)

Aveneid, 2023-12-15:
OpenOCD is in version 0.11.0.

Output of openocd command:
`root@aveneid-ThinkPad-T480s:/home/aveneid/Downloads/stm32f1-picopwner-main# openocd -f interface/stlink-v2.cfg -f target/stm32f1x.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Error: open failed`

Am I missing configuration? Sorry if I'm asking dumb questions.

CTXz, 2023-12-16:
What kind of debug probe are you using? Can you upload a picture of the probe attached to the bluepill?

Aveneid, 2023-12-16:
I forgot about the STLink.. Sorry for bothering you.
Now everything is working, need to adjust SRAM entry and should be good to go.
Thanks for help!

PS. Im using STLink v2 clone like this one: https://allegro.pl/oferta/st-link-v2-stm32-stm8-stlink-programator-swd-swim-9157100329

closed 2023-12-21 by CTXz