CTXz/stm32f1-picopwner #14: Strange SRAM entry point

SundaresanN, 2024-01-03:
Hi, i'm using stm32F103C8T6 blue pill. 

I'm getting the following error, 

    raise Exception("Could not determine SRAM entry point\nopenocd output: " + line)
Exception: Could not determine SRAM entry point
openocd output: xPSR: 0x01000000 pc: 0x08000150 msp: 00000000

I find it strange pc: 0x08000150. Its supposed to be 0x20000150. 

[image]

Additional info, I'm running bare metal led blink example with RDP enabled. Is it necessary to run any specific application to make the exploit work?

Thanks!

CTXz, 2024-01-24:
Based on the value of the `PC`, the device simply starts into flash instead of SRAM. This most likely implies that your boot pins are not being properly configured. I suspect `BOOT0` is the culprit in your case. Can you confirm it is correctly connected to the Pi's GPIO (GPIO5)? If you're confident, try pulling up the `BOOT0` pin via the jumper and see if the PC at least jumps into the SRAM. The exploit will obviously not work like this but it would confirm the Pi's not doing its duty of pulling the pin high.  If necessary, check out if the pin is being set by the pi using an oscilloscope.

CTXz, 2024-02-20:
Since this issue has been inactive for a prolonged period of time now, I'll be closing it. I'll gladly reopen it if necessary

closed 2024-02-20 by CTXz

Alex-com2, 2024-06-16:
Hello! When I try to read STM32F103RCT6 I get an error. In CubeProgrammer, the connection occurs (photo of registers attached). All Pins are connected correctly.
[image]
[image]