CTXz/stm32f1-picopwner #16: STM32F103

VR555, 2024-01-22:
Hi, what might be wrong, cycling this message

STDOUT:
SHELL: openocd -f interface/stlink.cfg -f target/stm32f1x.cfg -c init -c 'reset halt' -c targets -c exit
STDERR: xPack Open On-Chip Debugger 0.12.0+dev-01312-g18281b0c4-dirty (2023-09-04-22:31)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J43S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.426657
Info : [stm32f1x.cpu] Cortex-M3 r1p1 processor detected
Info : [stm32f1x.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
[stm32f1x.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x20000108 msp: 0x20005000
    TargetName         Type       Endian TapName            State      
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f1x.cpu       hla_target little stm32f1x.cpu       halted

STDOUT:
SHELL: openocd -f interface/stlink.cfg -f target/stm32f1x.cfg -c init -c 'reset halt' -c targets -c exit
STDERR: xPack Open On-Chip Debugger 0.12.0+dev-01312-g18281b0c4-dirty (2023-09-04-22:31)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J43S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.418283
Info : [stm32f1x.cpu] Cortex-M3 r1p1 processor detected
Info : [stm32f1x.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
[stm32f1x.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x20000108 msp: 0x20005000
    TargetName         Type       Endian TapName            State      
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f1x.cpu       hla_target little stm32f1x.cpu       halted

CTXz, 2024-01-24:
Appears to be simmlar to https://github.com/CTXz/stm32f1-picopwner/issues/15#issuecomment-1908447070. Please verify your BOOT1 pin is being pull up properly.

CTXz, 2024-02-20:
Since this issue has been inactive for a prolonged period of time now, I'll be closing it. I'll gladly reopen it if necessary.

closed 2024-02-20 by CTXz