CTXz/stm32f1-picopwner #17: another potential issue on the connection picture

gezzerlol, 2024-02-05:
connection picture doesn't imply a seperate debug probe to be used... I know it's a silly mistake to be done on the user side but I spent 2 hours on "waiting for a debug probe to be connected..." screen due to my sheer lack of knowledge. after telling my self "could this debug probe be an st-link?" I finally got this worked and dumped the firmware lol

CTXz, 2024-02-06:
Hi!

First off, glad to hear it still ended up working!

I do appriciate the suggestion but I'd like to mention that the README does mention in the "What you'll need to get started" section that a debug probe is required and even goes as far as to proivde the ST-LINK v2 as an example:

 > - A debug probe (e.g. a ST-Link V2)
 
It should also be noted that a debug probe musn't necessarily be a ST-Link, but may also be, say..., a J-link.

Further I'd like to point out this paragraph in the "Pre-Requisites" section:

> The instructions also require that you have a basic understanding of how to build Pico SDK based projects. **It also pre-supposes that you have a basic understanding of how to connect your** Pi Pico and **debug probe to your target STM32F1 board.**

Adding a debug probe to the illustration is also not necessary in my opinion as the probe is only used for the preparation of the exploit.

That being said, I propose as a solution to this potential missunderstanding that the `waiting for a debug probe to be connected...` message will be changed to `waiting for a debug probe (eg. ST-LINK) to be connected...`. I'm also considering to release a video that goes throught all steps required to test out the exploit on a bluepill.

If you have any disagreements and/or further suggestions, let me know!

gezzerlol, 2024-02-09:
You're right, I realized an st-link debug probe requirement is mentioned in the "What you'll need to get started" section, and later the message "waiting for a debug probe to be connected..." message explicitly mentions a debug probe, which is an st-link for most people. This is true for people who have the experience and understanding using a debug probe on an stm target, which you also listed as a prerequisite. But as a person who use st-link for debugging and programming but lacks near full understanding of terms and concepts, the term "debug probe" didn't click in my brain, since I don't use the name "debug probe" when mentioning an st-link device. The replacement message you proposed could make the tool more accessible to a few more people who are like me I believe. A complementary video would also be nice for people who are visual learners, if you bother to shoot one.

closed 2024-02-09 by gezzerlol