CTXz/stm32f1-picopwner #18: [Idea] Universal loader

deividAlfa, 2024-02-16:
I know it's notpretty ugly but this  should remove the need of making different loaders?

Place entryVec at 0x100:
```
SECTIONS
{
  /* The startup code goes first into FLASH */
  .isr_vector :
  {
   ...
    . = 0x100;
    *(.entryVec) /* entry vector position for SRAM execution */
  } >RAM
```

Fill ram 0x100-0x300 with jumps. Wherever the ram boots, it'll jump to the correct place.
```
.section .entryVec
    b _start
    b _start
    ... (256 times)
```

CTXz, 2024-02-20:
Implemented in https://github.com/CTXz/stm32f1-picopwner/commit/3bdbdd64eb0c30af236b515af399e9bb58c8f19d Will be merged into main along with some other changes.

I've only been able to test this on the F103, but I don't see any reason why this wouldn't work on other models. Obviously the trampoline has some overlap with the IVT, but since the affected interrupts aren't being used that ok I suppose.

closed 2024-02-20 by CTXz

deividAlfa, 2024-02-20:
Yep, shouldn't cause any issue, while bringing a lot  of benefits! 👍🏻