CTXz/stm32f1-picopwner #20: STM32F103C8T6 Serial dump does not stop

MoeBenMtl, 2024-03-01:
Hello I would like to thank you for sharing this utility and also ask a question about an issue I am having I have an STM32F103C8T6 Rev X 
the process works fine but when I get to the serial dumping part it keeps dumping data forever ... if anyone has an idea please let me know

MoeBenMtl, 2024-03-01:
I had the same reflection, I left it dumping until it reached 2048kb file size which made no sense.

CTXz, 2024-03-01:
I can't say for sure until I've tested, but perhaps it enters a loop where it resets after receiving the hardfault, which in return jumps back to the start of stage 1.

Try the following traget binaries and let me know if they fix the issue:

[target_firmware_v1.3.1tar.gz](https://github.com/CTXz/stm32f1-picopwner/files/14465007/target_firmware_v1.3.1tar.gz)

If this still doesn't do it, I'll take a closer look at this tommorow.

MoeBenMtl, 2024-03-01:
Same outcome loops forever past 128kb also does seem to be in some sort of loop as the data that's being printed looks repetitive

CTXz, 2024-03-01:
Alright, I'd just like to confirm that this is related to the latest release. Would you mind trying the last release?

https://github.com/CTXz/stm32f1-picopwner/releases/tag/a1.2_t1.2

Make sure to use the dump script from that release as well.

MoeBenMtl, 2024-03-01:
Yes sure NP

MoeBenMtl, 2024-03-01:
I wiped and used last release and I have the same exact issue.

MoeBenMtl, 2024-03-01:
[image]

here the dump

CTXz, 2024-03-01:
Huh, that's rather interesting. It appears that this issue is unique to your set-up then.

I'll see if I can implement a condition in the dump.py script tommorow that rejects incoming data after the connected device's flash size has been dumped. Or at the very least, I'll add a flag that can restrict the dump file size to a provided size. That way, even if such an endless loop occurs, the script will still complete the dump.

MoeBenMtl, 2024-03-01:
Alright, Ty also doesn't the data looks like it's dumping the same thing over and over ?

CTXz, 2024-03-02:
Wups, I didn't pay enough attention to the image. I had assumed it's looping at the stop of the flash. You're right, in this case the proposed solution obviously won't be very useful. It seems to be looping after transmitting 404 bytes. It's also clearly resetting, as the start magic `0x10ADDA7A` is being transmitted at the start of every repeat. Something must be triggering a reset.

I'll upload a couple of test firmware binaries later that could further diagnose what's potentially causing the issue. If you have an oscilloscope, perhaps you could monitor the reset line and see if it experiences some sort of voltage flunctiuations that could cause the reset to trigger. If that's not an option, a multimeter reading of the NRST pin could also potentially help. That being said, the reset being triggered at exactly 404 bytes makes a hardware related issue less likely.

MoeBenMtl, 2024-03-02:
You are right RST keeps getting triggered on original pcb nrst is tied to 3.3v
[image]

CTXz, 2024-03-03:
Ok, I've come up with a workaround that should be able to dump despite experiencing resets.

Please use the following attack and target fimrwares and let me know how things look this time:

[target_firmware_v1.3_rst_Workaround.tar.gz](https://github.com/CTXz/stm32f1-picopwner/files/14471463/target_firmware_v1.3_rst_Workaround.tar.gz)
[attack.zip](https://github.com/CTXz/stm32f1-picopwner/files/14471465/attack.zip)

The latest dump.py should be used

MoeBenMtl, 2024-03-03:
Amazing ! Ty will try in a bit and report

MoeBenMtl, 2024-03-03:
Omg ! This worked mint ! Thank you !

CTXz, 2024-03-03:
Sweet! In this case I'll close the issue now.

Cheers!

closed 2024-03-03 by CTXz

deividAlfa, 2024-03-23:
Regarding this:
I recently discovered clones/fakes report no flash size  (Flash size register=0xFFFFFFFF).
This will cause up to 64MB to be dumped, but shouldn't be a concern because accessing unimplemented memory should cause hard fault and inmediately stop the transfer.