CTXz/stm32f1-picopwner #23: STM32F4xx RDP 2

Mesteery, 2024-03-23:
I came across this project which looks promising and I was wondering if it was able to remove the write protection RDP 2 so stm32 can be flashed. Thanks

CTXz, 2024-03-27:
Hi, this exploit does not work on the STM32F4, let alone one with RDP2. If you wish to be updated about potential progress regarding the f4, the keep an eye on this thread: 

https://github.com/CTXz/stm32f1-picopwner/issues/13

closed 2024-03-27 by CTXz

deividAlfa, 2024-03-27:
RDP is irreversible

Angelic47, 2024-04-03:
Hi, the RDP2 can't be removed, even we hacked into the CPU and get arbitrary code execution - it's readonly, designed by ST for a security defense. 
In theory, the RDP2 can be temporary downgrade to RDP1, and for some combo glitches it could write the flash, implement a bootloader and makes it flashable again. 
But it's a quite challenging, due to the numerous security defenses filled by ST, makes it very hard to achieve. 
So downgrade the RDP 2 and reprogram the firmware will be worthless.