CTXz/stm32f1-picopwner #27: rp2040  zero is possible?

curcius, 2024-03-31:
I tried to execute the procedure with rap2040 zero but it is not working, the script is executed but it does not find rp2040. Is there anything that can be done to work on the rp2040 zero or does it only work on the pi pico?

``Script Version: 1.4
Requires Attack-Board Firmware Version: 1.x

Instructions:

1. Flash the attack firmware to the Pi Pico

2. Connect the Pi Pico to the STM32F1 target as follows (left Pico, right STM):
     GND     -> GND      
     GPIO 0  -> NO LONGER USED (previously USARTx_RX)
     GPIO 1  -> USARTx_TX
     GPIO 2  -> VDD      
     GPIO 4  -> NRST     
     GPIO 5  -> BOOT0    

Where:
     USART1_TX = PA9
     USART2_TX = PA2
     USART3_TX = PB10

3. Follow further instructions provided by this script
For more detailed steps, see the README.md file.

**Waiting for Pi Pico to be connected... (Looking for /dev/ttyACM0)**

curcius, 2024-03-31:
I also forgot to mention that it is a GD32F103 clone of the STM32F1, this uc is from an 8bitdo control that the analogues that are connected to the analogue ports do not work and 8bitdo does not provide the chip or the firmware to record another stm32f1.

Unfortunately I had to resort to the process of trying to dump the firmware to record on another stm32f103 so as not to lose my sn30 pro control. =(

[image]

curcius, 2024-03-31:
[image]

deividAlfa, 2024-03-31:
- Disconnect everything.
- Run `ls /dev/ttyACM*`.
- Connect pico.
- Run again `ls /dev/ttyACM*` and check if there's a new ACM port. It might not be ACM0, adjust arguments as required.

If exists and still not finding it, it might be a permission issue, try `sudo chmod 777 /dev/ttyACMx` (Adjust x as required)

CTXz, 2024-04-01:
In addition to @deividAlfa 's answer, see if `dmesg` reports anything when you connect the rp2040 board. It may provide a hint as to what's wrong.

curcius, 2024-04-01:
> * Disconnect everything.
> * Run `ls /dev/ttyACM*`.
> * Connect pico only.
> * Run again `ls /dev/ttyACM*` and check if there's a new ACM port.
>   It might not be ACM0, adjust `-p /dev/ttyACMx` argument as required.
> 
> If exists and still not finding it, it might be a permission issue, try `sudo chmod 777 /dev/ttyACMx` (Adjust x as required)

Thanks for responding, I actually forgot to check the permissions of /dev/ttyACM0, in my case I checked that it was the /dev/ttyACM0 port, I tested it by connecting and disconnecting the rp2040. I'm not home right now but when I get there I'll try chmod. Thank you very much

curcius, 2024-04-01:
> In addition to @deividAlfa 's answer, see if `dmesg` reports anything when you connect the rp2040 board. It may provide a hint as to what's wrong.

Excellent idea, I'll check in dmesg and set the permissions for /dev/ttyACM0, any news, I'll get back to you, thanks

deividAlfa, 2024-04-01:
In my case, I have to chmod it everytime I connect it. Not an issue as I only used it a few times for fun.
There must be away to modify the default permissions with rules or something.

curcius, 2024-04-01:
It has changed a little now the script asks to press the rp2040 reset button but it falls into the same condition as before, it does not continue with the exploit   =\

[image]

CTXz, 2024-04-01:
I suspect that the permissions for `/dev/ttyACM0` are reset when the pico is reset/replugged. A lazy way to fix this would be to try running the script with `sudo`. That should ultimately avoid any permission based errors from occuring.

deividAlfa, 2024-04-01:
You mistvrin the scriot before plugging the pico.
Connect when asked and chmod it.

curcius, 2024-04-01:
Now I went further, I installed Linux Mint and it was easier with two open terminals, one to run the script and the other to give permissions to the serial port.

But I still haven't gotten the dump. It is showing timeout as shown in the image.

I think the problem is related to the gpio power supply of the rp2040 for bluepill as per trobleshot:

The power draw of the target board is too high for the Pi Pico to handle (Try buffering the power pin with a BJT or MOSFET)

Does anyone have the connection diagram using the MOSFET?

Thank you for all your help =)

[image]

https://github.com/CTXz/stm32f1-picopwner/assets/1108722/8cf8e403-b739-4e82-b75d-2a80b20a654c

deividAlfa, 2024-04-02:
Don't connect ST-Link yo VDD pin.
Ensure to carefully follow the steps.
Don't work too fast, it will fail sometimes. Wait 2-3 seconds before proceeding to the next step.
But GD32 is not STM32,  the attach might not work.

curcius, 2024-04-02:
> Don't connect ST-Link to VDD pin. Ensure to carefully follow the steps. Don't work too fast, it will fail sometimes. Wait 2-3 seconds before proceeding to the next step. But GD32 is not STM32, the attack might not work.
> 
> @CTXz Suggestion: Use 2 or 3 gpios for VDD, so they can be joined more power if a single pin can't supply enough current.

I think the problem is incompatibility with **GD32f103** I have already done the entire procedure with great care but unfortunately I was not successful in the exploit. Now I have to accept that I lost control of 8bitdo =/

I greatly appreciate the support provided and wish everyone the best. Thank you very much.

banxian, 2024-04-04:
I think PWR led on blue pill board may shrink too much power.
gd32f103 is almost register level p2p replacement of stm32f103, but sometimes you may need tweakint RCC related (systeminit ) part in firmware.

deividAlfa, 2024-04-04:
I don't think so, my blue pill does just fine.
Measure VDD before and after "Press any key to start dumping the firmware" step.

RED4911, 2024-04-12:
[image]
Tell me, is there support for swclk, swdio?

curcius, 2024-04-14:
> ![IMG_20240412_223427](https://private-user-images.githubusercontent.com/6369947/322118173-3c900b8b-ae21-4748-a75c-d727d1af13b8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMwNjgzNDAsIm5iZiI6MTcxMzA2ODA0MCwicGF0aCI6Ii82MzY5OTQ3LzMyMjExODE3My0zYzkwMGI4Yi1hZTIxLTQ3NDgtYTc1Yy1kNzI3ZDFhZjEzYjguanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQxNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MTRUMDQxNDAwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDY2MjQ0MmU2NzViYjQzZTk1NzZjNTZiYzZkYzQyYTkzYzFiNTIwZTNiYzhhYmYyM2YyZTI1MDA5MGYwOTdmOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.S8iRNolbMiIo-dfxurOqI_6rTYsJSIJJ6Ty9U6AVtno) Diga-me, há suporte para swclk, swdio?

Is the stlink connected to the raspberry pi pico? Not in bluepill? Now it's confusing if it's to connect to the Raspberry, it really won't work because the rp2040 zero doesn't have these communication ports...

curcius, 2024-05-07:
[image]

I bought a raspberry pi pico according to the images on this github but it doesn't do the dump at all, I made all the connections according to the images in the readme but it's time to dump the timeout. Could anyone help? This GD32F103 damaged the analog ports that are connected to the analog stick of my 8bitdo controller and I wanted to remove the dump to record on another chip.

But I honestly don't know what I'm doing wrong because it always shows timeout as shown in the following image =\

deividAlfa, 2024-05-07:
Try waiting about 10 seconds between steps, I got similar issues by going too fast.
Anyways this is a stm32 attack, might not work in GD32.

curcius, 2024-05-07:
> Try waiting about 5-10 seconds between steps, I got similar issues by going too fast. Anyways this is a stm32 attack, might not work in GD32.

I was afraid that it wouldn't work on the GD32, I think I have an stm32, I have another bluepill board but I don't know if it has the stm32, I'll check and test with it if it works, then I actually lost my 8bitdo gamepad =/

tretek-prog, 2024-05-17:
> Боялся, что на GD32 не заработает, думаю у меня stm32, есть еще плата bluepill но не знаю, есть ли на ней stm32, проверю и протестирую на ней, работает ли, тогда Я вообще-то потерял свой геймпад 8bitdo =/

Hello, can I contact you by e-mail?

CTXz, 2024-05-17:
This issue is deviating from the original question and is becoming a discussion about whether the exploit works on the GD32. 

This topic deserves its own issue and should not be further discussed here.

closed 2024-05-17 by CTXz

curcius, 2024-05-25:
> > Боялся, что на GD32 не заработает, думаю у меня stm32, есть еще плата bluepill но не знаю, есть ли на ней stm32, проверю и протестирую на ней, работает ли, тогда Я вообще-то потерял свой геймпад 8bitdo =/
> 
> Hello, can I contact you by e-mail?

acurcius@gmail.com