CTXz/stm32f1-picopwner #31: BluePill: Timeout: No data received from target

eychei, 2024-05-18:
Hello,

I am trying to get this to work on a standard bluepill board without success.
Getting "Timeout: No data received from target" Error

What I tried so far:
-Using the precompiled firmware files
-Compiling attack and also target firmware
-Changing pcs and operation system: Windows WSL (Ubuntu), Virtualbox (Kali), Clean Linux PC (Ubuntu)
-Changed attack.c sleep.ms(15) to 30ms
-Changed Bluepill boards
-Changed PiPico W with PiPico
-Tried with unlocked stm32 firmware

Nothing of the above helped, always the same error.

What I have noticed is that the GP25 LED is not powered or blinking in any state of the process.
Hope someone can help me out in finding the root of this issue.

Thanks

eychei

curcius, 2024-05-25:
I have the same timeout problem, I couldn't solve it.

Straider777, 2024-09-18:
A similar situation, I tried on several stm32F1 chips, tried to cool the attacked board with alcohol cooled to -20 degrees. Getting "Timeout: No data received from target" Error. The supply voltage is supplied via a transistor switch, the pulse duration on the Vdd < 3 mcs, there are no capacitors.
[image]

Straider777, 2024-09-18:
Wow, alcohol cooling from the freezer, it still worked for the third time!
[image]
[image]
[image]

Julinho666, 2025-03-12:
What do you think the cooling effect has been?
The main article states: "The STM32F1 board is not genuine or perhaps too new (there are rumors that the exploit has been patched in 2020+ revisions of STM32F1 chips)." Perhaps that's why it doesn't work for many people?

Straider777, 2025-03-17:
In my opinion, this method does not work without cooling, I cannot explain anything about fixing the exploit in new chips, I had the task to read the firmware of one chip released before 2020. I didn't do this task anymore, as I solved the problem and repaired the faulty device.

Julinho666, 2025-03-19:
With my BluePill, it only worked with cooling (ice spray) at the beginning, then I removed all capacitors (including the reset pin) and after that it worked without cooling.

manter84, 2025-04-10:
Problem: No data received from target occurs only on BluePill. It does not occur when attacking other microcontrollers (not from BluePill). I removed the capacitors and cooled chip with alcohol, the attack did not work. I decided to just throw the board in the refrigerator and cooled it to -15C. The attack was successful. When the chip is heated to +15C, the attack is also successful. At +22, the attack does not work.  The attack works both on Linux (Ubuntu) and WSL2.

[image]

[image]

novval, 2025-06-18:
Hello everybody!
I have the same problem.
I also tried with bluepill. There is no original STM32F103. I checked the PA9 TX pin - it is constantly low. What could be the problem?
Is ShellCode not getting control? Or is the data in SRAM being corrupted? Or is it normal?

Julinho666, 2025-06-20:
You also removed U1, I see no cross?

novval, 2025-06-20:
No, I didn't delete U1 and C5, C7, C15, I'll try later and let you know the result.

manter84, 2025-12-04:
> manter84 можете поделиться файлами?И в stm32 записывали target?
Какими файлами поделиться? Зачем в стм записывать target? Ничего в стм записывать не нужно. Я делал все пошагово, как статье. Все получилось.

slualeksandr, 2025-12-05:
manter84 можете описать процедуру,как вы это делаете?

manter84, 2025-12-06:
> manter84,может быть я не правильно понимаю с помощью переводчика....Атака основана на том, что в оперативную память целевой платы STM32F1 временно загружается эксплойтная прошивка. Эта прошивка содержит двухэтапный эксплойт, который после завершения выгружает содержимое флэш-памяти целевой платы в последовательный порт

Я выше написал то что делаете непосредственно вы. Как подготавливаетесь к атаке. Как загружается эксплойтная прошивка в stm вы не видите. Точнее сказать это происходит в процессе атаки.

slualeksandr, 2025-12-06:
manter84,а зачем тогда используется st link?

slualeksandr, 2025-12-06:
manter84,в понедельник отвечу и опишу что и как..

manter84, 2025-12-09:
У вас скрипт не может определит rdp. Проблема либо в соединении (проверьте схему), либо  плохой контакт. Я напрямую подпаивался, без втычных разъемов.