CTXz/stm32f1-picopwner #32: Could GD32F103 support be possible ?

cxgth, 2024-06-16:
Hey, thanks for your work and implementation.

I'm facing the same problem as described in https://github.com/CTXz/stm32f1-picopwner/issues/27.
I tried to dump the firmware of an  STM32 clone (GD32F103).
Due to multiple failed attempts, I ordered a genuine STM32 blue pill and built a rig to be able to quickly swap microcrontrollers for testing.

The first attempt with a STM32F103 worked flawlessly.
So my guess is, right now it will not work with a GD32F103, but:

I found the CVE and original paper, the conference presentation @ WOOT '20 and another repository with an exploit for CKS and GD32 clones.

- [CVE-2020-13465](https://www.cvedetails.com/cve/CVE-2020-13465)
- [CVE-2020-13472](https://www.cvedetails.com/cve/CVE-2020-13472)
- [paper from CVE](https://www.usenix.org/system/files/woot20-paper-obermaier.pdf)
- [WOOT'20 conference](https://www.youtube.com/watch?v=20GghGAopXw)
- [sraptor_exploit](https://github.com/darkspr1te/sraptor_exploit)

The attack between STM32F103 and GD32F103 seems to be similar.
Do you think it could be implemented ?

yangzs001, 2024-06-22:
I also want to know how to extract the firmware of GD32F103

CTXz, 2024-06-24:
Thanks for the CVE references.

I'll take a look when I find the time. If they seem doable with an RP2040, I might get my hands on a GD32!

curcius, 2024-06-24:
I would be very happy if it were possible to dump the GD32f103 and thank you very much for your effort in making it work  @CTXz

chupalt, 2024-06-28:
GD32F103 doesn't require glitching
https://github.com/JohannesObermaier/f103-analysis.git

curcius, 2024-07-07:
> GD32F103 doesn't require glitching https://github.com/JohannesObermaier/f103-analysis.git

Do you have any tutorials?

curcius, 2024-07-07:
😭

dolphin22, 2024-07-15:
FPB doesn't work on GD32F103. D2 method from Johannes's research doesn't work for me as well.

Kiprus, 2024-07-29:
I tried using method D2 on the GD32F103, but only zeros were written to the RAM. The PC register has a value of 0, which suggests that there is no code execution, meaning the memory bus might be locked. I think the issue could be with the programmer or OpenOCD, as they might be accessing the core during initialization, which is causing us to get Debug_En.

dolphin22, 2024-07-29:
> I tried using method D2 on the GD32F103, but only zeros were written to the RAM. The PC register has a value of 0, which suggests that there is no code execution, meaning the memory bus might be [...]

I can load image to SRAM with BOOT0 and BOOT1 HIGH. I also facing the same issue with openocd 12-dev to load_image to flash, both with stlink and jlink. But it works with st-flash not sure why.

Kiprus, 2024-07-31:
> I can load image to SRAM with BOOT0 and BOOT1 HIGH. I also facing the same issue with openocd 12-dev to load_image to flash, both with stlink and jlink. But it works with st-flash not sure why.

Do you mean that you were able to read the firmware with RDP in this way?   Can you describe how you did this?

dolphin22, 2024-08-02:
> Do you mean that you were able to read the firmware with RDP in this way? Can you describe how you did this?

I am unable to dump flash firmware using this method because FPB doesn't work on GD32 families. I can load firmware to SRAM (BOOT0=BOOT1=HIGH) when security protection is enabled, see photo
[image]

curcius, 2024-10-31:
News about gd32? 🥹

dolphin22, 2025-02-16:
> News about gd32? 🥹

This guy used voltage glitches and successes. I haven't tried. https://www.youtube.com/watch?v=L1Qcnc3T99c