CTXz/stm32f1-picopwner #35: APM32F103CBT6

Testspektrum, 2024-12-17:
Is it possible to get firmware from APM32F103CBT6 when trying I get this problem:

Debug probe connected to STM32F1 target
Traceback (most recent call last):
  File "/home/osa/dump.py", line 422, in <module>
    rdp_status = get_rdp_status()
                 ^^^^^^^^^^^^^^^^
  File "/home/osa/dump.py", line 259, in get_rdp_status
    raise Exception(
Exception: Could not determine read protection status

raceback (most recent call last):
  File "/home/osa/dump.py", line 439, in <module>
    sram_entry_point = get_sram_entry_point()
                       ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/osa/dump.py", line 281, in get_sram_entry_point
    raise Exception("Could not determine SRAM entry point\nopenocd output: " + line)

Testspektrum, 2024-12-21:
It turns out that APM32F103CBT6 can be read without any problems :)
I tried several times but quickly.
I read to wait a few seconds after each step. That's what I did, with a minimum of 5 seconds between steps.
The microcontroller could be read without any problems by reading 128kb of code.
The reading is much slower than from the original STM32F103.

closed 2024-12-21 by Testspektrum

daohuaxiang4, 2025-01-24:
> 当我尝试遇到此问题时，是否可以从 APM32F103CBT6 获取固件：
> 
> ```
> Debug probe connected to STM32F1 target
> Traceback (most recent call last):
>   File "/home/osa/dump.py", line 422, in <module>
>     rdp_status = get_rdp_status()
>                  ^^^^^^^^^^^^^^^^
>   File "/home/osa/dump.py", line 259, in get_rdp_status
>     raise Exception(
> Exception: Could not determine read protection status
> 
> 
> 
> raceback (most recent call last):
>   File "/home/osa/dump.py", line 439, in <module>
>     sram_entry_point = get_sram_entry_point()
>                        ^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/osa/dump.py", line 281, in get_sram_entry_point
>     raise Exception("Could not determine SRAM entry point\nopenocd output: " + line)
> ```

How is this problem solved?

Testspektrum, 2025-01-24:
In my case, the APM32 was soldered onto a clean board without any capacitors that could cause problems. It was connected directly to the RP2040. At the beginning I had such errors, but as I wrote, after each step I waited at least 5 seconds until I performed the next step of the program. Finally, it worked and the layout was read without any problems. Later I tried several times and with this method I managed to read it many times.