CTXz/stm32f1-picopwner #36: Does it works on fake bluepill?

gcarmix, 2025-01-26:
I'm trying your PoC with a fake blue pill, using the connections that you showed.

However the dump.py seems not to work correctly, in particular  I had an error with openocd saying:

`Warn : UNEXPECTED idcode: 0x2ba01477
Error: expected 1 of 1: 0x1ba01477`

So I modified the stm32f1x.cfg of openocd to accept also the 0x2ba01477 cputapid.

Now i get the following error:

`Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.092034
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x1fffb1e8 msp: 0x20018400

Traceback (most recent call last):
  File "/home/ubuntu/stm32f1-picopwner/dump.py", line 440, in <module>
    sram_entry_point = get_sram_entry_point()
  File "/home/ubuntu/stm32f1-picopwner/dump.py", line 281, in get_sram_entry_point
    raise Exception("Could not determine SRAM entry point\nopenocd output: " + line)
Exception: Could not determine SRAM entry point
openocd output: xPSR: 0x01000000 pc: 0x1fffb1e8 msp: 0x20018400`

If it can help I noticed that the readout protection settings with this chip do not work correctly using STM32CubeProgrammer.
Using openocd command i can lock it, but I can't unlock it. The only software that set them correctly is the old ST-Link Utility for Windows.
This is the openocd output when trying to unlock:
`Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.085714
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x1fffb1e8 msp: 0x20018400
Info : device id = 0x20030410
Info : flash size = 128kbytes
Error: flash write algorithm aborted by target
Error: flash write failed at address 0x1ffff802
Error: flash memory write protected
stm32x failed to unlock device

shutdown command invoked`

ubuntu@ubuntu:~/stm32f1-pi

Do you have any clues?

deividAlfa, 2025-02-21:
Try skipping the RDP check by renoving this part:

https://github.com/CTXz/stm32f1-picopwner/blob/76f17499abf7691289889564730c8528f2212ca0/dump.py#L422-#L437

Julinho666, 2025-03-12:
Which programmer are you using? There are fake ST Link versions that don't work with STMCube, for example.

daohuaxiang4, 2025-03-13:
> 尝试通过更新此部分来跳过 RDP 检查：
> 
> [stm32f1-picopwner/dump.py](https://github.com/CTXz/stm32f1-picopwner/blob/76f17499abf7691289889564730c8528f2212ca0/dump.py#L422-L437)
> 
> 第 422 至 437 行 [76f1749](/CTXz/stm32f1-picopwner/commit/76f17499abf7691289889564730c8528f2212ca0)
> 
>   获取rdp_status状态 ​
>  如果 rdp_status 为：
>      print ( "确认STM32F1目标被读保护" )
>  别的：
>      print ( "STM32F1目标没有读保护，攻击可能没有必要" )
>      print ( "您是否仍想继续？ (y/n): " , end = "" )
>      而 True：
>          选择 = 输入().较低()
>          如果 choice  ==  "y"：
>              休息 
>          elif  choice  ==  “n”：
>              ser .关闭()
>              退出( 0 )
>          别的：
>              print ( "请回答‘y’或‘n’" )

> 尝试通过更新此部分来跳过 RDP 检查：
> 
> [stm32f1-picopwner/dump.py](https://github.com/CTXz/stm32f1-picopwner/blob/76f17499abf7691289889564730c8528f2212ca0/dump.py#L422-L437)
> 
> 第 422 至 437 行 [76f1749](/CTXz/stm32f1-picopwner/commit/76f17499abf7691289889564730c8528f2212ca0)
> 
>   获取rdp_status状态 ​
>  如果 rdp_status 为：
>      print ( "确认STM32F1目标被读保护" )
>  别的：
>      print ( "STM32F1目标没有读保护，攻击可能没有必要" )
>      print ( "您是否仍想继续？ (y/n): " , end = "" )
>      而 True：
>          选择 = 输入().较低()
>          如果 choice  ==  "y"：
>              休息 
>          elif  choice  ==  “n”：
>              ser .关闭()
>              退出( 0 )
>          别的：
>              print ( "请回答‘y’或‘n’" )

Please help me. I happened on apm32f103. Debug probe connected to STM32F1 target
Traceback (most recent call last):
  File "/home/osa/dump.py", line 422, in <module>
    rdp_status = get_rdp_status()
                 ^^^^^^^^^^^^^^^^
  File "/home/osa/dump.py", line 259, in get_rdp_status
    raise Exception(
Exception: Could not determine read protection status

gcarmix, 2025-03-16:
I bypassed the rdp as suggested, and as I reported earlier it then stops here:

```
Waiting for debug probe (e.g. ST-Link) to be connected...
Debug probe connected to STM32F1 target
Traceback (most recent call last):
  File "/home/ubuntu/stm32f1-picopwner/dump.py", line 439, in <module>
    sram_entry_point = get_sram_entry_point()
  File "/home/ubuntu/stm32f1-picopwner/dump.py", line 281, in get_sram_entry_point
    raise Exception("Could not determine SRAM entry point\nopenocd output: " + line)
Exception: Could not determine SRAM entry point
openocd output: xPSR: 0x01000000 pc: 0x1fffb1e8 msp: 0x20018400
```
The problem is that it looks for :  pc_str = "pc: 0x20000", while the pc value is 0x1fffb1e8.

Any clues?

deividAlfa, 2025-03-16:
0x1FFFxxxx is the internal bootloader, ensbled when BOOT0=1 and BOOT1=0.
Seems you're not pulling BOOT1 high, should be connected to VDD.

gcarmix, 2025-03-16:
I connected it exacly like shown in the picture of the readme file with the boot1 jumper in "1" position and boot0 wired to GP5 of the PICO. Could it be some problem of timing?

Julinho666, 2025-03-19:
Can you measure the Voltage on BOOT0 and on BOOT1. Both should be 3.3V, if the pico does not provide the voltage for BOOT 0 you will get the error.

gcarmix, 2025-03-28:
Also the voltage on both pins,measured with multimeter, are 3.3V