CTXz/stm32f1-picopwner #37: Does it work on a stm32f101 ?

RW2001, 2025-04-21:
I done all what need in faq and still nothing, what do i do wrong ?

[image]
[image]
[image]

CTXz, 2025-04-21:
The script says it:

```
Waiting for debug probe (e.g. ST-Link) to be connected...
```

You need to connect a debug probe to the STM board's SWD pins in order to upload the attack firmware into SRAM. One of those cheap knock-off programmers will do the job.

[image]

RW2001, 2025-04-22:
finally managed to dump, thank you for your work!

closed 2025-04-22 by RW2001

CTXz, 2025-04-22:
Nice, glad to hear you got it sorted out